EdgeTX/edgetx-sdcard-sounds #7: English voice changed - no longer clear and harder to distinguish

howels, 2021-08-17:
I flashed my radio using the Flasher tool but found that when I switched on the radio it was mumbling to me. The voice sounds like a breathy/soft anime voice and is very hard to make out words as the consonants are very soft. It's almost a whispering style of intonation and really hard to make out words.

Can we please revert to the original EdgeTX style of English voice? Radios don't have good speakers and I need to be able to distinguish the words easily.

nerdCopter, 2021-10-30:
sadly i agree.  i really wanted to give it a chance and benefit-of-doubt.
However, when piloting, some things are just hard to comprehend with the dialect/accent/pitch/volume of these sounds, especially numbers.

howels, 2021-10-31:
> sadly i agree. i really wanted to give it a chance and benefit-of-doubt. However, when piloting, some things are just hard to comprehend with the dialect/accent/pitch/volume of these sounds, especially numbers. We may need an "American" `US-EN`. man_shrugging

The original clips I have are using a more neutrally accented International/British English voice, and I don't find the American English accent is necessarily any clearer especially if the American accent is very strong. Are we using a speech synth or a real voice actor?  Simply adding a bit more emphasis to the phrasing would solve it. It's not the geography of the voice as much as the delivery of the words.

kaos-1, 2022-01-02:
We are also missing "Turtle Mode ON/OFF"

closed 2022-10-13 by pfeerick