EdgeTX/edgetx-sdcard-sounds #11: French sound pack

Daannoa, 2021-11-15:
Hello,

It is not the highest priority but,
In the french sound pack the hello.wav file says "bienvenu sur OpenTX" (=> "welcome to OpenTX") instead of ""bienvenu sur EdgeTX".

**General information**
 - EdgeTX version 2.5.0
 - Sound Pack version : 2.5.4 ( fr)
 - Hardware : Radiomaster TX16S

CoderElectronics, 2021-11-26:
@edernucci Might be able to fix this, i can try to make a correct modification and PR but my accuracy with french is below average haha

Daannoa, 2021-11-30:
Hello,

By me it is English which is below average! 😁 but if i can help let me know.

Thank you for your time

pfeerick, 2022-04-15:
This should be resolved by https://github.com/EdgeTX/edgetx-sdcard-sounds/commit/a2d59daef9a0a013775845573e526a017934cde4 and 857aa836a83ec99922ce4eb1cfd90574c954f43f

Please checkout the resulting audio and let me know if it needs further tweaking. 

https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/SOUNDS/fr/SYSTEM/hello.wav

closed 2022-04-15 by pfeerick

Daannoa, 2022-04-20:
Hello,
It is perfect like this! 
Thank you.