EdgeTX/edgetx-sdcard-sounds #15: Catalan Languaje

Xavitorradas, 2022-03-09:
I'd like to use Catalan languaje in my TX16. Meanwhile I have saved with my voice some sentences and changed them in 'es' sound folder and system ( not all changed an some else added ).
  How can I implement 'ca' languaje in EdgeTX ?
  Catalan is one of the 4 oficial languajes in Spain ( Castillian/Spanish, Catalan, Vasque and Galaic ) and has about 10 million of users in the world ( Also ANDORRA ).
  Thanks in advance
[voices.zip](https://github.com/EdgeTX/edgetx-sdcard-sounds/files/8215548/voices.zip)
.

pfeerick, 2022-04-06:
Would require changes both to the radio firmware to know about another language, as well as to be included here as another language pack.

Xavitorradas, 2022-04-12:
Meanwhile amb using ES folder with my catalan voice. Thanks

pfeerick, 2022-04-21:
You may also be interested in this when it merges ;) https://github.com/EdgeTX/edgetx-sdcard-sounds/pull/22

closed 2022-04-25 by pfeerick