EdgeTX/edgetx-sdcard-sounds #17: Finish translating additional words/phrases added

pfeerick, 2022-04-16:
The following phrases/words need to be translated into 

- cn / Chinese @//zyren Can you help with this?
- cz / Czech
- de / German @//rotorman Can you help with this?
- es / Spanish
- fr / French @//raphaelcoeffic Can you help with this?
- it / Italian
- pt / Portugese
- ru / Russian

```
    manmod.wav;manual mode
    acrmod.wav;acro mode
    angmod.wav;angle mode
    hznmod.wav;horizon mode
    lncmod.wav;launch mode
    airmod.wav;air mode
    pitmod.wav;pit mode
    navcrs.wav;navigation cruise
    navcsh.wav;navigation course hold
    navalh.wav;navigation altitude hold
    navpos.wav;navigation position hold
    navrth.wav;navigation return to home
    navwp.wav;navigation waypoint
    cruise.wav;cruise
    crshld.wav;course hold
    althld.wav;altitude hold
    poshld.wav;position hold
    rth.wav;return to home
    waypnt.wav;waypoint
    manual.wav;manual
    auto.wav;auto
    osd.wav;O S D
    osdprf.wav;O S D profile
```

pfeerick, 2022-04-16:
@rotorman Can you review this and see if it looks right?
handbetrieb
akro-modus
winkelmodus
horizont-modus
startmodus
luft-modus
grubenmodus
navigation kreuzfahrt
navigation kurs halten
halten der navigationshöhe
navigation positionserhalt
navigation zurück zum ausgangspunkt
navigation wegpunkt
kreuzfahrt
kurs halten
höhe halten
position halten
rückkehr zum heimatort
wegpunkt
manuell
automatisch
o s d
o s d profil
manuell
automatisch
O S D
O S D Profil

pfeerick, 2022-04-16:
@zyren Can you review this and/or make any changes? I'm not sure if 'Chinese (Simplified)' was the right dialect?
手动模式
角度模式
角度模式
地平线模式
发射模式
空气模式
坑道模式
导航巡航
航向保持
导航高度保持
导航位置保持
导航返回原点
导航航点
巡航
航向保持
高度保持
位置保持
返航
航点
手动
自动
O S D
O S D概况

pfeerick, 2022-04-16:
@raphaelcoeffic What say you? :grin:
mode manuel
mode acro
mode angle
mode horizon
mode lancement
mode air
mode fosse
navigation cruise
maintien du cap de navigation
maintien de l'altitude de navigation
maintien de la position de navigation
navigation retour à la maison
navigation waypoint
croisière
maintien du cap
maintien de l'altitude
maintien de la position
retour à la maison
point de cheminement
manuel
auto
O S D
profil O S D

pfeerick, 2022-04-16:
## cz / Czech

manuální režim
režim akro
režim úhlu
režim horizontu
režim startu
letecký režim
režim jámy
navigační plavba
udržování navigačního kurzu
navigační udržování výšky
udržování navigační polohy
navigační návrat domů
navigační bod cesty
plavba
udržování kurzu
udržování nadmořské výšky
udržování polohy
návrat do výchozího bodu
waypoint
manuální
auto
O S D
O S D profil

pfeerick, 2022-04-16:
## es / Spanish
modo manual
modo acro
modo ángulo
modo horizonte
modo de lanzamiento
modo aire
modo foso
navegación de crucero
mantener el rumbo de navegación
mantenimiento de la altitud de navegación
mantenimiento de la posición de navegación
navegación retorno a casa
navegación waypoint
crucero
mantenimiento del rumbo
mantenimiento de la altitud
mantenimiento de la posición
volver a casa
punto de ruta
manual
auto
O S D
Perfil O S D

Traducción realizada con la versión gratuita del traductor www.DeepL.com/Translator

pfeerick, 2022-04-16:
## it / Italian
modo manuale
modo acro
modo angolo
modo orizzonte
modo di lancio
modo aria
modo fossa
crociera di navigazione
navigazione mantenere la rotta
navigazione mantenimento dell'altitudine
navigazione mantenimento della posizione
navigazione ritorno a casa
navigazione waypoint
crociera
mantenere la rotta
mantenere l'altitudine
mantenere la posizione
ritorno a casa
waypoint
manuale
auto
O S D
profilo O S D

Tradotto con www.DeepL.com/Translator (versione gratuita)

pfeerick, 2022-04-16:
## pt / Portugese
modo manual
modo acro
modo angular
modo horizonte
modo de lançamento
modo ar
modo fosso
cruzeiro de navegação
curso de navegação
altitude de navegação porão
posição de navegação
navegação de regresso a casa
ponto de passagem de navegação
cruzeiro
curso de formação
altitude de retenção
posição mantida
regressar a casa
ponto de passagem
manual
auto
O S D
Perfil O S D

pfeerick, 2022-04-16:
## ru / Russian
ручной режим
акрорежим
угловой режим
режим горизонта
режим запуска
воздушный режим
режим ямы
навигационный крейсерский режим
удержание навигационного курса
удержание навигационной высоты
удержание навигационного положения
навигационный возврат в исходное положение
навигационная путевая точка
круиз
удержание курса
удержание высоты
удержание позиции
возврат к дому
путевая точка
вручную
авто
О С Д
O S D профиль

rotorman, 2022-04-16:
Best we ask also @wimalopaan and @gagarinlg to check as well, as I'm not a native German speaker, but below how it sounds OK to me:
```
manmod.wav;manual mode;"Handbetrieb"
acrmod.wav;acro mode;"Kunstflugmodus", one can also say "Akrobatikmodus", but I prefer Kunstflug, as more native IMO
angmod.wav;angle mode;"Winkelhaltemodus"
hznmod.wav;horizon mode;"Horizonthaltemodus"
lncmod.wav;launch mode;"Startmodus"
airmod.wav;air mode;"Luftmodus" (sounds awkward to me though, but I have no better alternative)
pitmod.wav;pit mode;"Boxenstoppmodus" oder "Wartungsmodus" (meaning maintenance mode)
navcrs.wav;navigation cruise;"Kreuzfahrt" (marine vehicle) or "Rundflug" for aerial vehicle
navcsh.wav;navigation course hold;"Navigation, Kurs halten"
navalh.wav;navigation altitude hold;"Halten der Navigationshöhe"
navpos.wav;navigation position hold;"Navigation, Positionserhalt"
navrth.wav;navigation return to home;"Navigation, zurück zum Ausgangspunkt" (means back to starting point, not home)
navwp.wav;navigation waypoint;"Navigation Wegpunkt"
cruise.wav;cruise;"Kreuzfahrt" (marine vehicle) or "Rundflug" for aerial vehicle
crshld.wav;course hold;"Kurs halten"
althld.wav;altitude hold;"Höhe halten"
poshld.wav;position hold;"Position halten"
rth.wav;return to home;"Zurück zum Ausgangspunkt" (back to starting point)
waypnt.wav;waypoint;"Wegpunkt"
manual.wav;manual;"Manuell"
auto.wav;auto;"Automatisch"
osd.wav;O S D;"Bildschirmanzeige"
osdprf.wav;O S D profile;"Bildschirmanzeigeprofil"
```

wimalopaan, 2022-04-16:
Winkelhaltemodus -> Winkelmodus: which angle? yaw, pitch, roll?
Horizonthaltemodus -> Horizontalmodus
Luftmodus -> Flugmodus
Navigation, Kurs halten -> Kurs halten
Halten der Navigationshöhe -> Höhe halten
Positionserhalt -> Positions halten

Kreuzfahrt: sounds funny to me, because "Kreuzfahrt" as a special meaning in german, but I've no better wording since the purpose is not fully clear to me

pfeerick, 2022-04-16:
Hm... I'm already starting to think need to move the translation of this to something better... I've been thinking for a while we should move some of the ETX stuff to crowdin ... maybe this is a change to see if it could work? While the output isn't directly usable in the current workflow, it is a CSV file, so won't be to hard to manipulate back into place. 

If you're willing to have a look, check out https://crowdin.com/project/edgetx-sdcard-sounds ... I've grabbed the US English as the source, let crowdin do the machine translation, and then added the UK English variant also. If anyone has any other suggestions I'm all ears ;)

brainbubblersbest, 2022-04-16:
@wimalopaan you dont fly Drones, Do you?
Give me a try, I aporeciate your Feedback. 😉
manmod.wav;manual mode;"Assistenzsysteme aus"
acrmod.wav;acro mode;"Steuerlimitierungen aus", 
angmod.wav;angle mode;"Autolevelmodus"
hznmod.wav;horizon mode;"Autolevelmodus Limitiert"
lncmod.wav;launch mode;"Startmodus"
airmod.wav;air mode;"Stabilisierungsmodus" 
pitmod.wav;pit mode;"Videosendeleistung reduziert" oder "Wartungsmodus" (meaning maintenance mode)
navcrs.wav;navigation cruise;"Autopilot halte Richtung 3D" (marine vehicle) or "Rundflug" for aerial vehicle
navcsh.wav;navigation course hold;"Autopilot halte Richtung"
navalh.wav;navigation altitude hold;"Autopilot Halte Position"
navpos.wav;navigation position hold;"Autopilot Halte Höhe"
navrth.wav;navigation return to home;"Autopilot, zurück zum Startpunkt" (means back to starting point, not home)
navwp.wav;navigation waypoint;"Autopilot starte Wegpunktmission"
cruise.wav;cruise;" "Assistierte manuelle Steuerung " 
crshld.wav;course hold;"Richtung halten" (Gyro/ACC) 
althld.wav;altitude hold;"Höhe halten" (nur Baro/Lidar) 
poshld.wav;position hold;"Position halten" (Opflow) 
rth.wav;return to home;"Zurück zum Startpunkt" (back to starting point)
waypnt.wav;waypoint;"Wegpunkt"
manual.wav;manual;"Manuell"
auto.wav;auto;"Automatisch"
osd.wav;O S D;"In Video Armaturen"
osdprf.wav;O S D profile;"VideoArmaturenprofil

brainbubblersbest, 2022-04-16:
This might be helpfull for better understanding. https://github.com/iNavFlight/inav/blob/master/docs/Settings.md

zyren, 2022-04-17:
> 

Yes I will be checking and improving these today.

LoakAudio, 2024-11-28:
@pfeerick 
years after, sorry.
In french fr_FR "mode fosse" is never used for "pit mode". We would say 'Pitmode activé" and "Pitmode désactivé"
Unlike Quebec Canadians, we do not hesitate to integrate English, particularly in technical areas.