EdgeTX/edgetx-sdcard-sounds #23: missing phrases

Meldinov, 2022-05-02:
i just realised the edgetx-sdcard-sounds/voices/es-ES.csv is missing many phrases.
And why there are two columns, "source text" and "translation", but both are in spanish?
I can translate to spanish, but i am pretty new to github and this code :P

pfeerick, 2022-05-03:
Because I only re-formatted the files yesterday, and they didn't have the source text heading in them before ;) Hold off doing any translation for a couple of days - I am in the process of getting this ready to use crowdin for handling the translation process better, hence the changes to the file format.

JaLav013, 2022-06-02:
Is there a "Turtle Mode On" Phrase?  I cannot find it to assign to a switch.  I have a new TX16S MkII, Version 2.6.0 with a date of 2/27/22 on it.

pfeerick, 2022-07-10:
> Is there a "Turtle Mode On" Phrase? I cannot find it to assign to a switch. I have a new TX16S MkII, Version 2.6.0 with a date of 2/27/22 on it.

Yes - you need to get the latest version of the voice pack. 

https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/voices/en-GB.csv#L331

msu2313, 2024-09-04:
I'm not sure if I didn't see it in the latest pack, but if not can "Loiter Change, Auto Level, Auto Tune" be added in the next release please?