EdgeTX/edgetx-sdcard-sounds #25: German voice pack not correctly

TheIsotopes, 2022-05-12:
According to the [de-DE.csv](https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/voices/de-DE.csv) file, 205 system and 26 other files are listed.
but in the zip file there are two files too many in the system folder, `engon.wav` and `fm-9.wav`, they don't belong there.

can someone please check that.

Meldinov, 2022-05-12:
the whole system is being remade. Will habe to complete and recheck the csv files and then, the new wav files can be produced

pfeerick, 2022-05-13:
That's because they are were somehow added in the [commit of files from the OTX voice pack](https://github.com/EdgeTX/edgetx-sdcard-sounds/commit/383345e8154825894cb3843b30780c4798e336ee)... I don't know how they got there though as they're not in the OTX pack. Eder was probably copying and pasting stuff around and got that one wrong. 

[image]

commit a9f92a0f (2022-05-13): Remove files that incorrectly added in 2bb3c49
Closes #25
closed 2022-05-13 by pfeerick