EdgeTX/edgetx-sdcard-sounds #34: Include sounds.json as a release asset

elecpower, 2022-08-06:
The file in the code directory reflects the pre-release build and not necessarily Latest or older releases.
This makes automating the installation of releases more challenging such as when using the sounds.json file for displaying available language packs for older releases. Then if chosen pack does not exist then handling the error situation.

pfeerick, 2022-08-11:
Sounds good... so just the file as part of the release, it doesn't need to be in the zips? I can retroactively add that to the say the last two releases then also.

elecpower, 2022-08-11:
Yes a separate asset file uncompressed.

pfeerick, 2022-08-11:
Ok, I've added to both the last release and pre-release. I'll make changes to the release action to finish this off properly so will close the issue through that once that's done.

elecpower, 2022-08-11:
Thanks. I'll mod the updates PR.

commit bfe1a2c4 (2022-08-11): chore: Include sounds.json as a release asset
Fixes #34
closed 2022-08-11 by pfeerick