EdgeTX/edgetx-sdcard-sounds #38: Inconsistent language strings in sounds.json

elecpower, 2022-08-25:
Mixture of hyphens and underscores which requires extra code when parsing in any automation.
eg "language": "cs_CZ" and "language": "de-DE"

pfeerick, 2022-08-26:
Yes, should be hypens - i.e. ISO style for the actual language portion, and then the underscore for voice variants.

Will also change the build folder names as I don't know why they were specified as `en_gb-voicevariant` when it really should have been `en-gb_voicevariant`. It'll affect the folder in the zip file, but since that is specified in the `sounds.json` it shouldn't be hard-coded anywhere.

pfeerick, 2022-08-26:
Ok, it's done in the repo, and I also last three releases to match, so Companion should like that now. 😉 I'll do the other changes after a PR goes in.

commit d8973cc8 (2022-08-26): fix: Consistently use ISO style lang strings
Resolves #38
closed 2022-08-26 by pfeerick