EdgeTX/edgetx-sdcard-sounds #42: Another VTX sound pack

SumolX, 2022-09-07:
I am going to make another sound pack related to VTX and LED settings.  Just to be clear no spacing for acronym based sounds.

Example: 
VTX Frequency R1
VTX Frequency R2
VTX Frequency R3
VTX Frequency R4
VTX Frequency R5
VTX Frequency R6
VTX Frequency R7
VTX Frequency L1
Etc....

LEDS Red
LEDS Green
LEDS Blue
LEDS Cyan
LEDS Yellow
LEDS White
LEDS Orange
LEDS Violet
LEDS Magenta
LEDS Off

I just want to be sure this will be OK before starting this effort.  I tested both real quick and sounded good.  It's too bad we can't combine multiple sounds together.

pfeerick, 2022-09-07:
> It's too bad we can't combine multiple sounds together.

Very much so... would be nice to be able to assemble individual words for custom phrases. Might be something we could explore in the future - might have to be more of a 'advanced feature' through a phrases config file or something, at least in the first iteration.

pfeerick, 2022-09-07:
lol... is that the one touch vtx/led on he just did... have to watch that one later.

> I have a related question is there a means for EDGETX to repeat all switch positions while in use.... similar to how it plays the switch positions during startup? It would be nice to push a momentary switch to give overall status of the switch positions while having the goggles on.

Not that immediately springs to mind. I do believe there is a PR will allow for this sort of behaviour - as someone else wanted this - basically another LS that could be used as part of a "current switch position" set of functions. I want to try and squeeze this in to 2.8 still... lets see how that goes 😆 https://github.com/EdgeTX/edgetx/pull/2085

closed 2022-09-10 by SumolX