EdgeTX/edgetx-sdcard-sounds #47: Swedish language not listed in sounds.json

ulfhedlund, 2022-09-22:
Swedish language is not listed in sounds.json, thus not possible to download/update in Companion.

ulfhedlund, 2022-09-22:
I suppose there might be more to it to be able to use Swedish sounds in the radio and Companion... 
It would be cool if this could be fixed for 2.8. If I can help I am willing to try (as a non software developer).

pfeerick, 2022-09-23:
There is no Danish audio pack, which is why it's not listed. I've just added `sv-SE` so it will be downloadable with Companion the next time I build the pack, which should be the next couple of days as I need to redo some of the English phrases - need some pauses. 

I can't make any promises for 2.8, it's late as it is. 😆

pfeerick, 2022-09-23:
Resolved by c37a04792122a815b6c9e682461af4757d525171

closed 2022-09-23 by pfeerick