EdgeTX/edgetx-sdcard-sounds #50: Missing phrase

CarlKnox, 2022-10-09:
I'm watching RCVR channel on YouTube where he's setting up FM1-FM3 to play two phrases each.
I don't know which voice pack he has, but he has a Sport Mode (sprtmd) phrase which I don't see on my SD (I do have the Sport phrase).
So, I checked GitHub and also don't see the sprtmd phrase under the EN folder.

I know Bardwell, on his channel, was saying there were a lot of phrases he had to record in order to create his voice pack. Is there an official list so I can check for other missing phrases? Might as well find out now so the keeper of phrases can commit them to the repository.

pfeerick, 2022-10-09:
John / RCVR is making a bunch of extra phrases and uploading them to his discord, which I would think he mentioned in the video/description. So he is probably using one of those, rather than the "stock" phrases. The lists of phrases are in the voices folder of this repo ;) => https://github.com/EdgeTX/edgetx-sdcard-sounds/tree/main/voices

CarlKnox, 2022-10-09:
I may have been watching one of his OpenTX videos. Sorry about the confusion.

closed 2022-10-13 by pfeerick