EdgeTX/edgetx-sdcard-sounds #52: Documentation

shkup, 2022-10-10:
Would you mind please mention what is the default EVN voice you use, so when we add phrases at the online demo we can use the same one?

pfeerick, 2022-10-10:
https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/generate.sh

No, it's all CSV files... look in the voices directory.

shkup, 2022-10-10:
Oh... So if I edit the sh with a supported  SDK voice (Michelle for example :) ) It should generate the sound package folder?

shkup, 2022-10-10:
Can you add it to the generate.sh before line 27? You have the environment all setup... It will generate it on the next nightly...

./voice-gen.py voices/en-US.csv en-US-MichelleNeural en_us-michelle || exit 1
./voice-gen.py voices/en-US_scripts.csv en-US-MichelleNeural en_us-michelle/SCRIPTS || exit 1

pfeerick, 2022-10-11:
There are no nighties for the voice packs, as that would result in immediate $$, especially when there are no changes, but I can certainly add that for the next build, for sure.

pfeerick, 2022-10-11:
Generated files were added in 7dd6c001abfabaf4a212646809f01454c62493d9, and post-processing and release job running now, so should be available in about an hour.

closed 2022-10-11 by pfeerick