EdgeTX/edgetx-sdcard-sounds #55: Documentation fix

shkup, 2022-10-15:
Please add a paragraph for custom phrases. 

For custom phrases there is no need for `ubuntu` or different installations. Phrases can be generated with `curl` or `http` client like `postman`.  
After building a text to speech resource in `Azure` you can use it by `REST` calls (`http` requests).  
The request url is:
`https://<YOUR_RESOURCE_REGION>.tts.speech.microsoft.com/cognitiveservices/v1`
You should add the following headers to your request: (EdgeTX supposed to support up to 32khz `.wav` file but in that range 24khz is the highest value supported by the conversion service),
```
Ocp-Apim-Subscription-Key: <YOUR_RESOURCE_KEY>
Content-Type: application/ssml+xml
X-Microsoft-OutputFormat: riff-24khz-16bit-mono-pcm
```
And in the request body (raw) place your `ssml` (Change the voice name according to your preference, the full list is: `tts.speech.microsoft.com/cognitiveservices/voices/list`):  

```
<speak version='1.0' xml:lang='en-US'>
    <voice xml:lang='en-US' xml:gender='Female' name='en-US-MichelleNeural'>YOUR_PHRASE_HERE</voice>
</speak>
```

Generally speaking, the implementation in `voice-gen.py` can be a series of `http` requests without the proxy objects. I don't see any benefit if they require ton of installations, but I might be wrong here. I'll be happy to hear if they contribute somehow.
It seems that it can be an `os` independent implementation.

pfeerick, 2022-10-15:
This is an enhancement, not a fix. What was documented was a known reliable mechanism, until someone like yourself pointed out alternatives ;)

re: `voice-gen.py` and also documentation generally, PRs are welcome. However, I have no intention of changing the current implementation as it mostly follows the suggested implementation from Microsoft, and is working fine as it is. I have no objections to a platform agnostic alternative being added also, but won't change the current implementation just to prove a point.

shkup, 2022-10-15:
I didn't say this implementation is not good or doesn't follow best practices. It is very well written. 
I was more curious about the difference.
Anyway, I think that documentation section with viable, relatively easy approach towards generating custom phrases is valuable. I really like adding special functions with the model that being loaded etc.
Thank you for your time and patience. Again, great work!

closed 2022-10-15 by pfeerick

pfeerick, 2022-10-15:
I honestly don't know the difference... I would expect it is mostly just a wrapper to make things more object/python like. 

You might also like the stuff that doesn't need special functions - like the model names, and switch audio based simply on a file being in the right folder... Can make life much easier if you don't change the switch functions around on the field, since you don't even need to add SF lines for the switch audio then. 

https://doc.open-tx.org/manual-for-opentx-2-2/advanced-features/audio#model-folder-sound-files

PR #56 (2022-10-15): Add documentation on programatic custom phrase generation
Resolves #55

credit: @shkup for the initial text as per #55