EdgeTX/edgetx-sdcard-sounds #62: Son Belges

BoulASA, 2022-12-02:
Serait-il possible de générer les sons suivants pour la Belgique et la Suisse ? Merci d'avance

.../fr/SYSTEM/
0070.wav --> Septante
0071.wav --> Septante et un
0072.wav --> Septante deux
0073.wav --> Septante trois
0074.wav --> Septante quatre
0075.wav --> Septante cinq
0076.wav --> Septante six
0077.wav --> Septante sept
0078.wav --> Septante huit
0079.wav --> Septante neuf

0090.wav --> Nonante
0091.wav --> Nonante et un
0092.wav --> Nonante deux
0093.wav --> Nonante trois
0094.wav --> Nonante quatre
0095.wav --> Nonante cinq
0096.wav --> Nonante six
0097.wav --> Nonante sept
0098.wav --> Nonante huit
0099.wav --> Nonante neuf

pfeerick, 2023-06-21:
> Would it be possible to generate the following sounds for Belgium and Switzerland? Thanks in advance

I'm not quite sure what you are actually after here... it seems like Belgium has three main languages - Dutch (Flemish), French, and German, and Switzerland uses either Swiss German or Standard German... 

Are you saying that the current version of 70-79 and 90-99 don't sound right for the French voice pack?

gonzalo-bulnes, 2024-01-02:
I know what @BoulASA meant here, it is a request to create new locales for the French language.

The numbers sound right for the `fr-FR` sound pack (French, locale France). However, in the `fr-BE` (Belgium) and `fr-CH` (Switzerland) locales, numbers in the 70-99 range are read differently.

Those locales are available in Azure Speech Services (e.g. `fr-BE-GerardNeural`, `fr-CH-ArianeNeural`).

pfeerick, 2024-01-03:
So does it require a separate CSV, or can the current [fr-FR.csv](https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/voices/fr-FR.csv) be used? Meaning just needs two more lines in [generate.sh](https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/generate.sh) to create the new voices, and [sounds.json](https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/sounds.json) updated.

gonzalo-bulnes, 2024-01-03:
I think it makes sense to create a new file, say, `fr-BE.csv`. This exactly a difference in locale (vs language) and the `fr-FR.csv` it correct in its own context (`FR`).

I am not aware of the differences between `fr-BE` and `fr-CH` that could apply to the current EdgeTx strings, so I'd start with one new file, and expand if someone asks for it.  (People in Belgium and Switzerland tend to sound different, but that's a _voice_ aspect not _locale_, so that wouldn't mecessarily require a new file.)

gonzalo-bulnes, 2024-01-04:
Wait, I just looked into adding the `fr-BE` file and realized I had missed something.

Since the numbers in the 70-99 range are specified as numbers (e.g. `70`, not `soixante-dix` or `septante`) then the `fr-BE` file would actually be identical to the `fr-FR` file, and it is the text-to-speech service that would make the difference. I'd expect that picking a voice for the `fr-BE` locale like `fr-BE-GerardNeural` will read `70` a `septante`.

I still think it makes sense to have a file per locale, because they are different locales, but the _smallest_ acceptable change in this case seems to be to generate sounds with a `fr-BE` voice from the `fr-FR.csv` file.

```bash
./voice-gen.py voices/fr-FR.csv fr-BE-GerardNeural fr || exit 1
```