EdgeTX/edgetx-sdcard-sounds #63: Directory for Swedish sound pack should be /se, not /sv

SBWings, 2023-01-01:
Someone on a social media page was having problems with the error "No data files" when attempting to change their TX16S to the Swedish language. He posted pictures where he was attempting to create a Special Function. I downloaded a couple of language packs to experiment. They all worked except for Swedish. I successfully tried German /de, French /fr and of course, my native language English /en. When I tried the Swedish language pack /sv, I got the same error message. I searched for the correct country abbreviation for Sweden and it it is SE, not SV. I renamed the directory on my SD card from /sounds/sv to /sounds/se and it worked! I am now able to switch between English, German, French and Swedish on my radio.

I request that you rename the Swedish sound pack in the repository from edgetx-sdcard-sounds-sv-2.8.0.zip to edgetx-sdcard-sounds-se-2.8.0.zip and make the corresponding changes to the file itself so it extracts into a correctly named directory.

Thanks for all your wonderful work on this project! People do appreciate what you are accomplishing!

pfeerick, 2023-01-03:
SV is actually the correct designator per one of the ISO extended sets that is used to also indicate regional languages, but you are right - the radio is expecting to see `SE`, so will need to update the builds and index to reflect that. Thanks for the report! :)

[image]

SBWings, 2023-01-03:
Hi Peter,
Thanks for the quick response! I found SE when I googled the 2-character country code abbreviation. I just tried it again, this time asking what country is SV. It came back with El Salvador. Here is the link. I guess it depends on which source or maybe it's because I'm querying from the US. I can't explain the difference. 

sv country abbreviation - Google Search

| 
| 
|  | 
sv country abbreviation - Google Search

 |

 |

 |

This issue came up on the EdgeTX Users Facebook pages. Martin Samuelsson posted that he was frustrated after a year of trying to fix it. He said my solution worked for him. However, I will post an update that you are working on a fix and that in the future, /sounds/sv will work as expected.

I used to be a computer programmer so this piqued my analytic curiosity. I used the same skills to also avoid deleting the model8.yml situation while still fixing the problem. I studied the code at Line 302 and I saw the null value in the track filename. On my model8.yml file, I just changed it to a known wav file, like acro.wav. That fixed it for me and I was still able to keep the file. 

I wish I could help with EdgeTX but most of my experience was with procedural languages like COBOL I did take a class in MS Visual BASIC but that was in the 1990's and I taught myself to hand-code HTML in the same time period. 

Thanks again!
Steven Beatty

pfeerick, 2023-01-03:
While you're messaging him, can you ask if the current 2.8.0 works now as is, as I also retroactively modified the current release file ;) It is also possible that a 2.8.1 Companion release will be needed when it comes out soon, as there were some fixes to some translation stuff there also. 

It's more which standard you go by... the ISO has two letter, three letter, numeric, and now are working on a four-letter variant (so not confusing at all), and in this case, I'm actually using[ ICU Unicode locale descriptors](https://icu.unicode.org/), so `sv-SE` means "[Swedish (Sweden)](https://www.localeplanet.com/icu/sv-SE/index.html)" as opposed to `sv_FI` for "[Swedish (Finland)](https://www.localeplanet.com/icu/sv-FI/index.html)". 

Yeah, the `model8.yml` issue was annoying - that model config that RM shipped on the TX16S as standard was corrupt as it was made using a pre-release firmware. The value should have been quoted, and then it being blank wouldn't have been an issue. The other way to fix that sort of issue is to let the transmitter firmware fix the file itself - i.e. open the model, make a simple change, and then reverse that change. In doing so, the complete model file is re-written and any invalid values are either corrected or removed. 

Any help, whether it be code, reporting bugs, following up on bug reports to make sure they get resolved, or helping others is greatly appreciated. 

Should be resolved going forward by 5a40d2bc9ded837b8219d50519ae7c566775cbc0

closed 2023-01-03 by pfeerick

SBWings, 2023-01-03:
Thank you again, Sir!

 
 

Closed #63 as completed.