EdgeTX/edgetx-sdcard-sounds #70: request sounds for Camera Control

jawz101, 2023-06-22:
I have a Runcam Thumb Pro which supports connection to a flight controller to start & stop recording video via Betaflight's support of the Runcam protocol to control cameras.  There are flight modes available in Betaflight to start recording, configure the camera, or enable WiFi on a camera (the thumb doesn't have WiFi but I believe others do.)

Request sounds to say things like:
"video/recording/camera start/begin"

"camera settings or configuration"

[image]

pfeerick, 2023-06-23:
Will something like this do? Limitd to 6 leter filenames atm for non-system sounds, so hopefully these make sense.
```csv
"String ID","Source text","Translation","Context","Path","Filename"
"443","Camera wifi button","Camera wifi button","camera","","cwifi.wav"
"444","Camera power button","Camera power button","camera","","cpower.wav"
"445","Camera change mode","Camera change mode","camera","","cmode.wav"
"444","Camera start","Camera start","camera","","cstart.wav"
"445","Camera stop","Camera stop","camera","","cstop.wav"
```

jawz101, 2023-06-23:
Those are great.
I might also consider additional non-specific phrases such as "recording started" "recording stopped"
That could be applicable to blackbox recording or video recording (considering I'd also requested blackbox recording phrases it makes me think generic phrasing might apply to users' various scenarios.

PR #71 (2023-07-03): Add camera control phrases
Resolves #70
commit bdc38351 (2023-07-03): feat: Add camera control phrases (#71)

* Add camera control phrases
Resolves #70

* typo

* Add "Recording start" and "Recording stop"

* Add "start recording" and "stop recording"

* Somebody can't count

* Add StringID duplicate build check
closed 2023-07-03 by pfeerick