EdgeTX/edgetx-sdcard-sounds #73: release builder should strip `v` from version tag

pfeerick, 2023-07-03:
Version tags have the `v` present, but it seems like from past releases this hasn't been included in the filename. Until I've ascertained the collateral damage to Buddy and Companion that would result from leaving this, `release.sh` should strip it out.

gonzalo-bulnes, 2024-01-02:
Hi @pfeerick, is this still an issue?

Subsidiary question: is it safe to assume that the interface with Buddy and Companion is restricted to the GitHub releases page?

pfeerick, 2024-01-03:
As far as I know - I've not looked any further into this. And yes, just via the releases API.

gonzalo-bulnes, 2024-01-05:
Checking if I'm understanding the issue correctly, and if I'm reading the code correctly:

- The assets from the [last release](https://github.com/EdgeTX/edgetx-sdcard-sounds/releases/tag/v2.9.0) don't include the `v` in their names (e.g. `edgetx-sdcard-sounds-cz-2.9.0.zip`)
- The current `release.sh` script does create ZIP files with the content of the `VERSION` variable in their names
- The `VERSION` variable does contains a prefix, e.g. `v2.9.0` because the tags contain it ([code](https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/3093192f9f8aa2328cfc1f59d10bfc005ff33d17/.github/workflows/build_release.yml#L34-L35), [tag](https://github.com/EdgeTX/edgetx-sdcard-sounds/releases/tag/v2.9.0))

... does that mean that the last release wasn't created by the release workflow? (Or that the asset names were edited before upload?)

pfeerick, 2024-01-05:
Releases can be edited and files renamed after release ;) so you are
reading it correctly, the names were edited after release to correct for
the aforementioned deviation from prior behaviour.

PR #93 (2024-01-06): fix: Remove 'v' prefix from release VERSION
Fixes #73

Details in the commit message.
closed 2024-01-06 by pfeerick