EdgeTX/edgetx-sdcard-sounds #81: There is not Ukrainian language in a voice language menu

kotanidze, 2023-10-14:
There is not Ukrainian language in a voice language menu

josser, 2023-11-15:
Same for me

pfeerick, 2023-11-19:
The main firmware will need something similar to https://github.com/EdgeTX/edgetx/pull/3511 and https://github.com/EdgeTX/edgetx/pull/3586 but for Ukrainian

This is best done by a native language speaker due to idiosyncracies in various languages and dialects. Or is the construction of the Ukrainian language similar enough to that of Russian that I could use as a base, and you guys could proof it to see if it is saying the right things at the right time?

It would also be nice at that point if a native language speaker could also work on translating the firmware, but that is another task in itself.

pfeerick, 2023-11-19:
i.e. if you wanted to test that... just rename the 'ua' sound pack folder to 'ru', and select 'russian' as the voice language, and see how that works ;)

josser, 2023-11-19:
@pfeerick tested with ru, works fine. But I can test not all phrases :(

pfeerick, 2023-11-20:
Can elaborate more as to ` I can test not all phrases`? If you just mean you can't check every single word/phrase, that's fine... the events in particular to check for would probably be ones involving numbers (i.e. timer countdown, play value readout, does it say hundred/ hundreds, thousand/thousands correctly and at the right time, etc). And from the RU text to speech handling, it seems like there is some difference between male and female voice? Anyway, I'll throw together a PR soon for UA and link it here so you can give feedback.

josser, 2023-11-20:
@pfeerick oh, sorry, it was some combination of issue with autocorrection and not good English )

Yes. You are right, I've tested ability to enable language. It says welcome phrase, but I'm just novice in FPV and edgetx overall, so I need some time to understand how to trigger phrases with numbers.

pfeerick, 2023-12-13:
@kotanidze @josser Please give this PR a try, and give us feedback in that PR. https://github.com/EdgeTX/edgetx/pull/4415

I will close this issue now as this is a firmware issue that should be discussed in that PR. 

> To access the automatic builds for a PR, first open up the PR page. Under the PR title, you should see a series of tabs (Comments, Commits, Checks, Files Changed). Click on the Checks tab. Then if you are after the radio firmware, click on the link on the left side mentioning firmware. If you want a Companion/Simulator build, pick the one that suits your operating system. Then, under the 'Artifacts' heading near the bottom of the page will be a link to the zip file containing either the firmware or companion package.

closed 2023-12-13 by pfeerick