EdgeTX/edgetx-sdcard-sounds #88: Developer docs: a few questions to get started contributing to this repository (engineering, with first-hand localization experience)

gonzalo-bulnes, 2024-01-02:
Hi @pfeerick (and everybody else who can answers these questions!)

I'm interested in contributing some thinking and tooling to this repository. 
I see opportunities to reduce duplication, automate further and make the job easier for people interested in providing or correcting sounds in their own language. But I'd like to make sure I understand how this repo relates to the other EdgeTX projects before making the wrong assumptions.

Can I check my understanding of the goals with you?

1. This repository releases include:
   - one archive per language-locale-voice, with the contents of the `/SOUNDS/<language code>` directory of the EdgeTX SD Card.
   - one `sounds.json` file
   
 2. Who consumes `sounds.json`? I don't see it used in the SD Card, is it retrieved from the releases page by Companion or Buddy?
 
 3. Regarding locales and voices: my understanding is that EdgeTX is not currently aware of the locale and voice, only the language.
    By that I mean: I can place the contents of the `us_GB-libby` archive in my SD Card's `/SOUNDS/en` directory and the contents of `es_CL-Catalina` in `/SOUNDS/es`. That will allow me to switch the voice language from `English` to `Spanish` from the radio settings, but I can't have `en_US-Ryan` at the same time, unless I hack my way around (e.g. my placing the contents of the archive in `/SOUNDS/de` and selecting `German`. Is my understanding correct?
 
 4. The files in `/SOUNDS/SYSTEM` are directly referenced by EdgeTX code, and as such are **required** for a language pack to be complete. But the files directly at the root of `/SOUNDS` are not, and can be considered optional. Is that true?
 
 5. Similarly, the files in `/SOUNDS/INAV` are **required** for the INAV language pack to be complete, but the script does not rely on any file outside of that directory. Is that also true?
 
 6. Companion and Buddy rely on the GitHub **Releases** page directly to retrieve language packs and information about them, so the content of the *Assets** section can be thought of as the interface of this directory with Companion and Buddy. Is that correct (I'm half-guessing here.)
 
7. There MUST be an archive named `edgetx-sdcard-sounds-<language code>-<edgetx version>.zip` for each of the supported languages for Companion and Buddy to function as expected. But the archives that follow a different naming pattern (e.g. `edgetx-sdcard-sounds-es_cl-catalina-2.9.0.zip`) are only meant to be downloaded by individuals who visit the **Releases** page. Is that correct? (Again, I'm guessing here.)

pfeerick, 2024-01-03:
2. Yes, Companion uses that to identify the packs and where they go. I believe Buddy does also. 
3. Correct, the firmware currently only recognises as single two letter folder, and specific ones that that. One of the long term goals is to remove that requirement. 
4. `/SOUNDS/SYSTEM` is used by the firmware itself, `/SOUNDS` are the user presented list of sounds, available for use by special functions, etc. 
5. This was added in order to allow the end user to replace voices included with iNav, BetaFlight and/or YAAPU telemetry scripts, for consistency with the rest of the radio phases and sounds. It is entirely up to the developers of those scripts if they use them natively or if end user has to replace the files as needed. 
6. Yes, the releases page is the interface for Companion/Buddy
7. No. The releases page is the sole place the archives are exposed, so Companion must use a combination of the `sounds.json` and the list of files present in a given release. 
8. For the firmware, have a look at the tts_##.cpp files in https://github.com/EdgeTX/edgetx/tree/main/radio/src/translations
For Companion, have a look at https://github.com/EdgeTX/edgetx/blob/main/companion/src/updates/updatesounds.cpp

Feel free to make any suggestions/changes to the `edgetx-sdcard-sounds` structuring that help with moving towards something like crowdin (for machine translation and ease of non-developer translator access) and doesn't completely tie us into a specific TTS service. Anything that touches on the firmware/buddy/companion will take longer, as the current system, while not optimal, works, so isn't as high a priority.

gonzalo-bulnes, 2024-01-04:
I'll archive/close this question thread BTW!

closed 2024-01-04 by gonzalo-bulnes