EdgeTX/edgetx-sdcard-sounds #94: build_release action - deprecated node12

pfeerick, 2024-01-06:
`.github/workflows/build_release.yml`

Uses `marvinpinto/action-automatic-releases` which is dependent on node12, which is now depreciated. Need to swap it out for another upload/release action. It is also responsible the two `set-output` warnings, which will also stop working soon.

pfeerick, 2024-01-06:
One possibility is to switch to https://github.com/tj-actions/git-cliff for the changelog, and https://github.com/ncipollo/release-action for the upload action. Both are actively maintained actions. 

example stub combining the two
```
name: Release

on:
  push:
    tags:
      - 'v*.*.*'

jobs:
  release:
    name: Release
    runs-on: ubuntu-latest
    steps:
      - name: Checkout code
        uses: actions/checkout@v4
        with:
          fetch-depth: 0 # Important for changelog

      - name: Generate changelog with git-cliff
        uses: tj-actions/git-cliff@v1
        with:
          args: --latest --strip all
          output: "CHANGELOG.md"

      - name: Create Release
        uses: ncipollo/release-action@v1
        with:
          artifacts: "dist/*"
          token: ${{ secrets.GITHUB_TOKEN }}
          bodyFile: "CHANGELOG.md"
          draft: false
          prerelease: false
```

gonzalo-bulnes, 2024-01-06:
I haven't looked yet, just a thought: would it be an option to update to `marvinpinto/action-automatic-releases` upstream?

pfeerick, 2024-01-06:
Not a chance... it has basically been abandoned as it hasn't been updated in three years. Hence there are a lot of open issues for it.  Better to change to something it is actively supported.

commit 7906d820 (2024-03-25): chore(ci): replace release action

old action was no longer maintained. Resolves #94
closed 2024-03-25 by pfeerick