EdgeTX/edgetx-sdcard-sounds #105: Bad French sounds

pierrotm777, 2024-03-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Bad translated sounds, example: 0010.was return two hundred and not one hundred and one.

### Expected Behavior

I have found that the from from 00101.wav to 00189.wav have a bad translated sound.

### Steps To Reproduce

play sound are bad

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

w11

pfeerick, 2024-03-01:
This sounds (pun not intended) like it is a sound pack issue not a firmware issue, hence why moving this to the sdcard-sounds repo. 

Or perhaps an understanding issue?

> 0010.was return two hundred and not one hundred and one.

Do you mean 0101.wav? If so, that is correct, it is *supposed* to be 200. 

https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/fd7454639f47aa49c0d1836f54ca067b343b8928/voices/fr-FR.csv#L103

0100.wav to 0109.wav count from 100 to 1000 in increments of one hundred. If the system needed to speak a number like 109 it would play the 0100.wav and 0009.wav files (not sure if a joining 'and' sound is played or not, it would probably be language specific). 

With respect to the actual source text used to generate the wav files, what is the acutal issue here? Note the second column "Source text" should be in English, but isn't yet, as this pretty much the same source file that was used with OpenTX. 

https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/voices/fr-FR.csv

pierrotm777, 2024-03-01:
Ok, if i understand you use the ID number for define wav name.

Sorry, so it's not an error !

pfeerick, 2024-03-06:
Yes, pretty much ;) Since they are in the SYSTEM folder, they're not pickable for playback via special functions, so normally no-one would notice that anyway. But this is the only practical way to do this... you only need the numbers from 1 to 99,  each of the hundreds to one thousand, and 'AND' , then you can conceivably count all the way up from 1 to 1999 with only 111 sound files.

closed 2024-03-06 by pfeerick