EdgeTX/edgetx-sdcard-sounds #111: Add real voice to language/locale: uk_UA

p3ngu19z, 2024-05-15:
## Description

I [found](https://www.instagram.com/tv/CgCRgPQI5OJ/?utm_source=ig_web_copy_link&igsh=MzRlODBiNWFlZA==) [(google drive)](https://drive.google.com/drive/folders/1qzFfbYw4vBnAh9kmXtYUbzmgwg7lizD0 ) high quality sound pack from real voice for Ukraine language.

Can I add this pack to Voice packs? 

## Help make EdgeTX better!

- [ ] (optional) I can help reviewing the new voice packs.

  [gallery]: https://speech.microsoft.com/portal/voicegallery
  [voices]: https://learn.microsoft.com/en-us/azure/ai-services/speech-service/language-support?tabs=tts#prebuilt-neural-voices

pfeerick, 2024-05-16:
Will need the permission of the author, or a (GPLv2 compatible) license that allows it to be redistributed before can add just any random audio to the voice pack.

inventor7777, 2024-05-30:
You can also convert the files to the correct WAV PCM 32000hz settings and add them to your radio yourself.
@p3ngu19z

closed 2026-09-04 by pfeerick