EdgeTX/edgetx-sdcard-sounds #112: Paus after words in German language file is to long

Gernod, 2024-05-20:
The pause after the words in German language are too long. 

Example: Battery Voltage is pronounced: "ten         volts"  should be "ten volts"

pfeerick, 2024-05-30:
If you have a long pause between words, try reloading the SD card pack, as you probably have the original un-trimmed audio from the repo, rather than the trimmed and normalised version that is in the releases. 

i.e. this is the audio from the repo for "ten" and "volt"... would run for nearly 3 seconds, with quite a delay after each file. 

[image]

Whereas the audio in the release zip files is clipped much shorter, as well as normalised so it is louder, and would take less than half the time to play, with a much shorter delay after each file. 
[image]

Gernod, 2024-05-30:
Indeed, you are right. Reloaded the right soundpack and its working. Thank you.

pfeerick, 2024-05-30:
Glad that fixed it. Enjoy! :)

closed 2024-05-30 by pfeerick