EdgeTX/edgetx-sdcard-sounds #116: Additional surface phrases.

inventor7777, 2024-06-05:
Hello again!

I’ve been making a list while using my radio, and I have some more phrase suggestions. 

They are:

“Drift Mode Enabled”
“Drift Mode Disabled” _(I know there are ones that sound similar, more on that in a sec)_
“Screenshot”
“Cell” _(for including in prompts about cell voltage)_
“Transmitter Battery Low”
“Left”
“Right”
“Shift”
“Stability Control On”
“Stability Control Off”
“Stability Control Enabled”
“Stability Control Disabled”
“Thermal Warning” _(thermal and warning already exist, so it would be there just to make the trigger sequence for the phrase simpler)_

Now, as I’ve been selecting phrases, I’ve noticed something interesting: some phrases have an On/Off end, and some have Enabled/Disabled or even Active instead (see phrases like “course hold…” or “rescue mode…”. So I’ve added some phrases like “drift mode enabled/disabled” to pair well with other lines like “wet mode enabled/disabled”

in the future it’s probably something to look out for, and to ensure integrated, smooth phrase transitions.

pfeerick, 2024-06-06:
>  or even Active instead

They were probably deliberately done that way so the last word _wasn't_ similar, so less of a chance of mishearing which one it said. This is why there ends up being a couple of different versions of them though, to suit preferences. 

“Transmitter Battery Low” is already present as a system file... you want to be able to play that via special function rather than let the radio manage when it should be played?

Rest all makes sense though.

inventor7777, 2024-06-06:
> you want to be able to play that via special function rather than let the radio manage when it should be played?

Exactly. I just want the option for it to warn me once at, say, 6.8v, so I have plenty of time to charge before it starts vibrating and yelling at me 😂.

> They were probably deliberately done that way so the last word wasn't similar, so less of a chance of mishearing which one it said. This is why there ends up being a couple of different versions of them though, to suit preferences.

As for Active, I totally get it (since when does "launch control enabled" sound cool). I just think for some files both options (enabled/disabled and on/off) should be available.

PR #117 (2024-06-10): feat: add more surface phrases
Closes #116
closed 2024-06-10 by pfeerick