EdgeTX/edgetx-sdcard-sounds #119: Add language/locale: En_Ireland

stevempotter, 2024-07-11:
## Description

I'd like to be able to CREATE A voice pack in the following language(s) and locale(s): English in an Irish accent.

This is not a request to use AI to generate a new dialect, but to get help on how my wife, who is Irish, can record a voice pack of the correct format. She does some voiceover work and has good clear enunciation in a fairly neutral Dublin accent.

Thanks!

- `en_IRL` 

## Help make EdgeTX better!

- [ x] (optional) I can help reviewing the translations.
- [ x] (optional) I can help reviewing the sounds once they're generated.

pfeerick, 2024-07-16:
We are using a somewhat eclectic mix of two letter ISO 3166 country code and and ISO 639 two letter locale code, so that would be en-IR, for English, Ireland, but that sounds fine and would be great. Some competition for "Emily" :laughing:

stevempotter, 2024-07-16:
Great. Can you help with instructions?

pfeerick, 2024-07-18:
Basically you will want another entry in `sounds.json` like this one...
https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/46ced91e122d99006cefbb7e743fb509ced63fd0/sounds.json#L32-L37

And audio files for all the entries in 
/voices/en-GB.csv @ https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/voices/en-GB.csv  - 
which would go in a new folder in 
/SOUNDS @ https://github.com/EdgeTX/edgetx-sdcard-sounds/tree/main/SOUNDS

perhaps un-creatively named `en_ir` unless you want to ... name... the pack somewhat... in that case, it will just need the en_ prefix. 

Bonus points if you can convince her to do /voices/en-GB_scripts.csv @ https://github.com/EdgeTX/edgetx-sdcard-sounds/blob/main/voices/en-GB_scripts.csv also. 

If you want to do words/phrases not in /voices/en-GB.csv, that is fine... just need to add them (and /voices/en-US.csv also as that is basically a copy at this point). 

Hopefully the release script does not mangle the audio when it tries to clip any trailing silence / normalise the audio, else I'll need to teach it some new tricks so it leaves that alone. Audio files need to be PCM Wave files, no higher than 32khz sample rate, mono.

stevempotter, 2024-07-22:
Thanks.

#Hopefully the release script does not mangle the audio when it tries to clip any trailing silence / normalise the audio,

I hope this audio processing (what is a release script?) is something you will do? Or am I responsible for that? If so, I will need some hand-holding.

Looking forward to en_ir-mairead on my TX!

pfeerick, 2024-07-26:
Yeah, that will be my problem to deal with. 

It's the `release.sh` script that is in the repo root directory, that does all the audio trimming, normalisation and builds the zip files for each sound pack. Whereas `generate.sh` builds all the wav files via Azure TTS that are in the SOUNDS directory.

closed 2026-09-04 by pfeerick