EdgeTX/edgetx-sdcard-sounds #132: Add words or phrases (Surface)

inventor7777, 2025-01-15:
Hi! I have a couple suggestions for some new surface phrases that could be useful for crawlers.

**“Drag Brake”** _(to combine with other phrases for example)_
**“Drag Brake Off”
“Drag Brake On”
“Drag Brake Low”
“Drag Brake Medium”
“Drag Brake High”**
_(The last three are optional, as we already have standalone low, medium, and high phrases, but it sounds better if it’s in one file. I honestly don’t care very much, but it depends on what the future plans for simplifying the sounds library looks like)._

one other thing that I’ve noticed is that in some voices (such as US-Guy) Gyro is pronounced as _“euro”_ which isn’t helpful unless radios are being used at currency exchanges 😂. In other voices it says _“gyroscope”_ which is rather long. At this point I’m wondering if it’s even needed, as the AI voices don’t seem to understand that we just want _“gyro”_, especially as we have _“stability control”_ phrases already. Just pointing that out.

inventor7777, 2025-03-14:
@pfeerick any word on these? Thanks 😊

inventor7777, 2025-03-20:
TBH I never even thought of that….lol

pfeerick, 2025-03-20:
Have a look at #137... doing the voice gen (second commit) is easy for me as just have to run the script, so don't worry about that. But if if you think you're up to the first part of the PR, I'm more likely to either complain that I don't like it, or just generate the audio and merge 😉

inventor7777, 2025-03-20:
@pfeerick that looks pretty simple….just need to change the generate CSVs (as far as I can see…)

I’ll probably try later tomorrow (here in the US it’s almost 21:00) 

I don’t think I’m quite ready to attempt to actually generate the sound files, but maybe one day I’ll get Linux running in a VM and try to do it.

inventor7777, 2025-03-20:
How do phrases get added to other languages? Do sounds get added to EN first, then eventually get translated?

pfeerick, 2025-03-20:
Ideally they would get added to all languages at the same time, but the third column, "translation" would be left blank until someone translates it. Just had to double check the generator does actually skip over untranslated entries 🤭

PR #140 (2025-03-21): feat: add new surface phrases
@pfeerick I've taken your advice to try and make a PR and here it is!

I wasn't really sure if, or how, I needed to name the commits so I just left it as standard. If I need to change it, let me know, or maybe you can do it before merging. I've done my best to keep the file naming standard and all have 6 characters or less.

This PR adds
**”Track Mode”
“Drag Brake” 
“Drag Brake Off”
“Drag Brake On”
“Drag Brake Low”
“Drag Brake Medium”
“Drag Brake High”**
but no sounds have been generated.

I did notice that you opened a PR to regenerate sounds, hopefully this won' [... cut]
closed 2025-03-21 by pfeerick