EdgeTX/edgetx-sdcard-sounds #136: Building my own voice files sound different from official sound pack

psfales, 2025-03-19:
This is more of a request for information than a bug report.  Feel free to close it if that's not appropriate, or redirect me to more appropriate channels.

I've set up my build environment so that I can recreate the voice files using "generate.sh" and "voice-gen.py"
It seems to mostly work, and generates voice files that sound like they are spoken by the right person, but my files don't sound quite the same as the ones in the official sound pack because the cadence and accentuation is different.   

I've only done a small amount of experimenting so far, mostly with manmod.wav in the SOUNDS/en directory.    To my ear, the official file sounds like  "manualmode" with no special emphasis.    But the one I generate sounds more like "MANual mode" spoken more slowly and with an emphasized first syllable.)

If it's just me, then maybe it's no big deal.   But if something has changed on the Microsoft Azure side, maybe it's something that folks might want to investigate and understand.    

If it helps, I've attached a zip containing two files.  manmod.wav is the one from the standard sound pack in SOUNDS/en/manmod.wav.    manmod-newbuild.wav is the one I get when I run generate.sh.   (I tried to just attach those file directly, but it didn't allow .wav file type.   The .zip file was allowed) 

[manmod.zip](https://github.com/user-attachments/files/19350078/manmod.zip)

pfeerick, 2025-03-19:
Have a look at the post-processing done in `release.sh` ... the files are clipped and normalised there which could explain part or most of the difference. The first in the repo are the initial audio, and then post processed when the release packs are built.

psfales, 2025-03-19:
Thanks for that clarification - I understand the process better now, but I claim that the files I posted earlier are (somewhat unwittingly!) comparing apples to apples.   In the zipfile I posted eariler, manmod.wav is the file from downloaded source tree, which is the as-built (un-clipped, and un-normalized) file, and manmod-newbuild.wav is the newly built file (also un-clipped and un-normalized).   They are clearly different (both audibly, and visually in a tool like Audacity, where the newly built one is a bit longer)

pfeerick, 2025-03-20:
Thanks for confirming. 

Hm, yes, something does appear to have changed upstream in the last three years since that audio was generated 😆 

[image]

To my ear the enunciation is better, and the words are less crammed together... so looks like it is finally time to rebuild the lot.

psfales, 2025-03-20:
Thanks for looking into it.   I like the new one better as well!    Just puzzled as to why my build was different.

psfales, 2025-03-20:
It looks like if you rebuild the files it will help with another of my longstanding annoyances, namely the way Emily pronounces "Volt"  (it sounds to me like "vault")!

pfeerick, 2025-03-21:
This should be resolved by #139, but please do let me know if anything seems "off" :)

closed 2025-03-21 by pfeerick